dronekit/droneshare #144: droneshare is unable to analyse binlogs written by ArduCopter 3.2.1-rc2

hsteinhaus, 2015-02-01:
Here is an example: http://www.droneshare.com/mission/58044/plot
The raw binlog file is here:  https://www.dropbox.com/s/8xwat0uwejvahl4/68.BIN?dl=0

BTW, LogAnalyser.py throws an exception, too. After converting the mission to text format, the LogAnalyser seems to work, but produces funny output (complains about 20000% mag field change and offsets of 65k), while all plots of the flight look pretty reasonable).

jasonmartens, 2015-07-15:
Hello @hsteinhaus, I tried downloading your log, but it seems the link isn't working anymore. We did have a bug in LogAnalyzer which was resolved here: https://github.com/diydrones/ardupilot/pull/2541. Maybe that fixed your issue?

hsteinhaus, 2015-08-06:
I fixed the problems (at least for the linked logs) with https://github.com/diydrones/ardupilot/pull/2241

Unfortunately I did not preserve the linked logs to verify the fix again, but I just tried a recent log written by ArduCopter 3.3-rc7 - without any issues.

closed 2015-08-06 by hsteinhaus