dronekit/droneshare #145: Cannot modify profile

DustyAir83, 2015-02-19:
My droneshare username is DustyAir83 and I am unable to modify my profile.  I would like to be able to add my email address an a picture for my screen title.

mrpollo, 2015-02-19:
@geeksville can you help @DustyAir83 please?

DustyAir83, 2015-02-19:
I am no longer able to log in.

DustyAir83, 2015-02-19:
also, when I click "forgot password" I get the response 
"No email address known for this account, sorry."

DustyAir83, 2015-02-22:
I'd also like to add that my password is shorter than the required 7 characters.  I believe that i was an early adopter and set my password before there was a 7 character minimum.  In any case.  I am now no longer able to log in, and since I don't have an email address associated with my account yet I can't change my password.  Is there anything I can do other than create a new account?  I'd really like to keep my original screen name.

Dcphoto, 2015-04-06:
I am also not able to add an email or add a photo avitar as well. It would also be cool to have more fields in our profile.

kaitlynhova, 2015-07-15:
*We'll check into this.

Quick answer on Profile photo for now: DroneShare is pulling your photo from your (gravatar)[https://en.gravatar.com/] account. If you want to change it you have to got to Gravatar to take care of it.