dronekit/droneshare #146: Logs with Error Packets get 'Invalid Location Data' Error

krbaker, 2015-02-22:
I have a number of logs (from 3.2 arduopilot) which contain Error packets:
Example
'\xa3\x95\x13\x06\x01'

These get an error about invalid location data on upload.  This seems like it should upload and maybe list this as some sort of error... invalid location seems not right?

krbaker, 2015-02-22:
From pymavlink it thinks this is ERR {Subsys : 6, ECode : 1}

RedPhoton, 2015-03-05:
I think I'm seeing the same thing, a lot. (How do you filter the log file)?

Here is dropbox link to one of my log files, https://www.dropbox.com/l/rBSFlKf8Yy0onHI5h5AyNq

krbaker, 2015-03-08:
Hi,
  Took me a bit to get back to this.  I'm using a script I wrote that uses pymavlink to throw away the ERR packets in the mavlink log.  You'll need to install pymavlink and then set it up properly.  Once its setup you should be able to run my script.  Its not pulled into the mavlink repo yet, but I'm going to submit a pull request there.
Mavlink toolset: https://github.com/mavlink/mavlink
My Addition: https://github.com/krbaker/mavlink/blob/master/pymavlink/tools/mav_filter
  Keith

RedPhoton, 2015-03-08:
Amazingly I was just playing around with mavlink when you posted. This is very helpful, thanks!

visser, 2015-05-10:
DS can't scan the mavlink log and ignore that line? Seems crazy we need to edit log files prior to submitting, this is also breaking mavlink log uploads within MP. :|

twenty8special, 2015-05-18:
I am having the same issue of "Invalid location data was found in that log, ignoring" error on the website, as well as errors when uploading via APM Planner.  I went through my log file and deleted the ERR sections I could find but that hasn't helped.  I uploaded my log file to the link below.  

Thanks

http://www.flyingtrashcan.com/crash2.log

peterbarker, 2015-07-14:
In regards to that crash2.log - it is malformed.  The FMT format - the first row - is lacking "Columns":

--- /home/pbarker/crash2.log.orig	2015-07-14 13:53:59.092625605 +1000
+++ /home/pbarker/crash2.log	2015-07-14 13:54:14.569072422 +1000
@@ -1,4 +1,4 @@
-FMT, 128, 89, FMT, BBnNZ, Type,Length,Name,Format
+FMT, 128, 89, FMT, BBnNZ, Type,Length,Name,Format,Columns
 FMT, 138, 25, AHR2, IccCfLL, TimeMS,Roll,Pitch,Yaw,Alt,Lat,Lng
 FMT, 1, 23, ATT, IccccCCCC, TimeMS,DesRoll,Roll,DesPitch,Pitch,DesYaw,Yaw,ErrRP,ErrYaw
 FMT, 136, 21, BARO, Iffcf, TimeMS,Alt,Press,Temp,CRt

Have you any idea how that might have happened?

Peter

FlyingTrashcan, 2015-07-14:
No idea.  I just imported the BIN file into APM planner and then exported the LOG file.  Could the autopilot not have not been initialized or setup properly initially??

peterbarker, 2015-07-14:
Unlikely it was the autopilot incorrectly set up.

Could you post the original .BIN file somewhere for me, please?

FlyingTrashcan, 2015-07-14:
I uploaded my last 4 BIN files.  The one that you have the log file from is:
http://www.flyingtrashcan.com/45.BIN
There are also these other 3 if you would like to see some others.   
http://www.flyingtrashcan.com/44.BIN
http://www.flyingtrashcan.com/43.BIN
http://www.flyingtrashcan.com/42.BIN

peterbarker, 2015-07-14:
The FMT header problem appears to be a problem in APM Planner 2.

https://github.com/diydrones/apm_planner/pull/765

There is a second problem with your text-format dataflash file - the GPS messages are all missing the last two entries:

FMT, 130, 45, GPS, BIHBcLLeeEefI, Status,TimeMS,Week,NSats,HDop,Lat,Lng,RelAlt,A
lt,Spd,GCrs,VZ,T
GPS, 3, 597234800, 1844, 9, 2.08, 33.3865368, -111.9127126, 0, 367.19, 0.13, 226.25

The dump I made with APM Planner 2.0.18-rc2 does not have this problem:

GPS, 3, 597245800, 1844, 9, 2.08, 33.3865359, -111.9127107, 0.33, 365.49, 0.08, 226.25, 0.09, 83922

What version of APM Planner did you use to make the text-version of the dataflash file?

Thanks,

Peter

FlyingTrashcan, 2015-07-14:
I am using 2.0.17.  That is the newest version I can see for Mac.