dronekit/droneshare #148: Please put something about "what this project is" in the tagline/readme/welcome letter/etc.

nfriedly, 2015-03-27:
Hey, someone sent me a link to this and, trying to figure out what it is, I looked at the tagline which just says "The new version of droneshare - built upon DroneAPI" ... not very helpful. Next I hit the readme - same thing, but with a link to the welcome letter. So then I click through to the welcome letter and...  info on contributing, but nothing about what the project is.

mrpollo, 2015-03-27:
Nathan, DroneShare is a shared repository of drone missions, It was conceived by @geeksville with the idea of having an easier way of sharing logs, and it eventually grew into the current state, the backend (server behind this) is called **DroneKit-Cloud** is part of **[DroneKit](http://dronekit.io/)** by [3D Robotics](http://3drobotics.com/).

I hope this small brief helps :)

nfriedly, 2015-03-27:
Profusely! PR coming in a moment ;)

closed 2015-03-27 by mrpollo