dronekit/droneshare #156: Unable to manually upload logs without valid GPS data

lvale, 2015-05-22:
I was trying to upload some indoor testing logs (without GPS) but the manual page http://www.droneshare.com/vehicle/731

throws out an error of :

Invalid location data was found in that log, ignoring

Shouldn't it allow to parse this kind of logs without GPS data ??

lvale, 2015-05-26:
Also APM Planner 2.0.17 is unable to upload indoor flights, because of the location issue
[image]

AndKe, 2015-06-30:
I have examples of normal flights where the message is shown for no apparent reason - logs does have valid GPS fix at all time.

mrpollo, 2015-10-13:
thanks @lvale @AndKe I think we already identified the issue and will be deploying a fix shortly, as soon as that happen I'll update this issue

closed 2017-01-02 by lvale