dronekit/droneshare #159: Invalid location data was found in that log, ignoring

hyperion11, 2015-09-29:
Can't upload log. 
Error: Invalid location data was found in that log, ignoring
Link to my log
https://www.dropbox.com/s/d9bqa3fnn1zau4z/560941b2a68980bb376ed700.log?dl=0

rmackay9, 2015-10-13:
I've found the same issue with all Copter-3.3 logs

mrpollo, 2015-10-13:
thanks @hyperion11, @rmackay9 this seems to be a different issue from what we have originally found for 3.3 (TimeMS vs TimeUS), I'm reviewing.

hyperion11, 2015-10-14:
@mrpollo, if manually edit log and delete all lines with "err" log successfully uploading. copter 3.2.1.
May be automatically reject lines with errors while uploading?

peterbarker, 2015-10-14:
The log is malformed.  Here's the format:

FMT, 19, 5, ERR, BB, Subsys,ECode

here's an example ERR line:

ERR, 16, 2

What's missing there is a sequence number:
CTUN, 542046, 640, 12, 652, 0, 0.38, 0.35, 0.00, 0.00, -250, 51
ATT, 542106, -17.94, -11.56, 0.00, 2.35, 30.24, 33.38, 0.19, 0.78

That second column is a sequence number - missing from ATT.

Where did this log come from?

rmackay9, 2015-10-14:
In 3.2.1 the ERR message didn't have a time on the front perhaps.  That shouldn't cause the whole upload to fail though I think.

mpjf01, 2015-10-16:
I get the same message with all log files
This is the shortest I have - change .gif to .bin
.[image]

rmackay9, 2015-10-16:
@peterbarker has fixed the LogAnalyzer.py script which I think was the main blocker for DroneShare not working.  I think @mrpollo needs to incorporate that script change into Droneshare now.

mrpollo, 2015-10-16:
Sadly the LogAnalyzer.py isn't the only culprit here, we have some dependencies we need to fix in order to be able fix the main problem which seems to be some data flash log syntax changes, the code is tightly coupled to the messages old format.

project in question is here: https://github.com/dronekit/arduleader

imrj, 2015-12-08:
is this going to be fixed any time soon? its a shame the site has become unusable after AC release 3.3.x