dronekit/droneshare #160: Contact email not working

AndKe, 2015-09-29:
Tried to notify admin of most features that stopped loading few days ago :

Delivery to the following recipient failed permanently:

     support@droneshare.com

Technical details of permanent failure:

DNS Error: Address resolution of mxa.mailgun.com. failed: Domain name not found

----- Original message -----

DKIM-Signature: v=1; a=rsa-sha256; c=relaxed/relaxed;
        d=gmail.com; s=20120113;
        h=mime-version:date:message-id:subject:from:to:content-type
         :content-transfer-encoding;
        bh=18OYSDGPaiBYMBoPIllKzmlmk3Y2reFdvnbuCw24A/Q=;
        b=P4Wet3TJxcDkuh+ym5+s1iN2uCEOyhov5pdzTo0gCKoo34HvjESBDt68BJRJv4Cvp0
         wkBoX9tLtY24M0zf+hXJAMJZNWWFnkiCfqmWDp2rKr/T7b9dmgg1Jf63091x6kD7SDn9
         7kvFmcwcZhbvveNW1fK1CV0HYHfE1KVTRlfO4PPolfPyrk2ZjWpxhkEmGc2+izMKIpq3
         JkiGhAsgXBo/LA9pzswNj4FyadfrAEukqtBKAq6za4b6jPvdQAb15mXUaCIe7NPtPnKT
         T5MajAcrPqcXrznM6Qbnv6voTkInzVItwpHpYUv2iGUU46XmNIUZM7BuGqwAm0IUmjlz
         oyDQ==
MIME-Version: 1.0
X-Received: by 10.55.214.5 with SMTP id t5mr14107616qki.66.1442696349348; Sat,
 19 Sep 2015 13:59:09 -0700 (PDT)
Received: by 10.140.100.228 with HTTP; Sat, 19 Sep 2015 13:59:09 -0700 (PDT)
Date: Sat, 19 Sep 2015 22:59:09 +0200
Message-ID: <CAA4d5Xn5yEmG8cFoK4yy1SjYX0ztkoe06uVAkcVfChvusuUH6A@mail.gmail.com>
Subject: The page is barely moving,
From: =?UTF-8?Q?Andr=C3=A9_Kjellstrup?= <andre.kjellstrup@gmail.com>
To: support@droneshare.com
Content-Type: text/plain; charset=UTF-8
Content-Transfer-Encoding: quoted-printable

mrpollo, 2015-10-13:
thanks, we are aware and trying to deploy a fix

LeonDBL, 2016-02-15:
I see the support email still isn't working after nearly 5 months, surely a fairly basic fix?

hamishwillee, 2016-02-16:
@LeonDBL The plan going forward is to accept support queries from the [dronekit cloud support forums only](https://discuss.dronekit.io/c/cloud). I am closing this issue.

closed 2016-02-16 by hamishwillee