dronekit/droneshare #163: developer site fail

xros, 2015-10-13:
I am following the instruction from the [welcome.md](https://github.com/dronekit/droneshare/blob/master/WELCOME.md#this-is-droneshare---please-join-us)

https://developer.3drobotics.com/
But the site for developers failed. So in this case I can not register a developer key. :cry: 

Can someone fix it? 

Please let me know if the site's back to life :smile:

xros, 2015-10-13:
I found another site
it's http://dronekit.io/support
guess it may be an alternative of the site above.

mrpollo, 2015-10-13:
hey @xros you can sign up for a key here https://cloud.dronekit.io/, thanks for reporting this bug

xros, 2015-10-13:
@mrpollo thanks mate.
But now I am having trouble of doing so.

I followed the tutorial from here.https://cloud.dronekit.io/docs#show-all-users

But that returned a HTTP 403 error.
It said my appkey was invalid. But actually I have the correct appkey from here  https://cloud.dronekit.io/ . 

Neither it works if I changed it with appid for instead.

Do you know what wrong that is?

mrpollo, 2015-10-13:
hey @xros can you post about your cloud usage problems at our forums https://discuss.dronekit.io/c/cloud ?, I'm sure more people might have a similar question, also more people can help answer

xros, 2015-10-13:
@mrpollo thanks. I went to the site there. But I found I can not register membership using github account.  Does its github OAuth authentication fail?